SystemRDL/PeakRDL #1: RALBot command line tool development

amykyta3, 2020-04-05:


zhajio1988, 2020-04-24:
hello @amykyta3 
I alreay have a script do the same job, but not use yaml config file. It will use files under a directory then generate files which you need based on switch option supply.
Below is the script help:
usage: ralbotgen.py [-h] [-i FILENAME | -merge | -show] [-inc_dir <dir>]
                    [-t <addrmap>] [-header [{all,verilog,c}]]
                    [-uvmregs [{ordt,in_house}]] [-doc] [-verilog] [-xml]
                    [-all] [-s] [-w <warning/no-warning>] [-v] [--debug]

optional arguments:
  -h, --help            show this help message and exit
  -i FILENAME, -input FILENAME
                        input systemRDL file name.
  -merge                merge rdl files imported in pico.rdl then generate.
  -show                 show all availabile systemRDL file name.
  -inc_dir <dir>        Include search path.
  -t <addrmap>, --top <addrmap>
                        Explicitly choose which addrmap in the root namespace
                        will be the top-level component. If unset, the last
                        addrmap defined will be chosen
  -header [{all,verilog,c}]
                        generate systemverilog or C header file(default all of
                        them).
  -uvmregs [{ordt,in_house}]
                        generate UVM reg model.
  -doc                  generate register html documents.
  -verilog              generate register verilog module.
  -xml                  generate IP-XACT xml file.
  -all                  generate all kinds of files.
  -s, --skip-not-present
                        If set, compiler skips nodes whose \u2018ispresent\u2019
                        property is set to False.
  -w <warning/no-warning>
                        Enable warnings (-w warning-name) or disable (-w no-
                        warning-name)
  -v, --verbose         Enable verbose compilation status print out.
  --debug               Enable compiler debug mode (with yet more compiler
                        status print out).

EngRaff92, 2021-11-21:
Hello just modified the JSON exporter to parse Memories, the json is then fed into another py script which will generate a list of SV parameters used in TB/RTL.

Should I paste code here ?
Regards

gdavis-github, 2022-06-15:
I would be willing to work towards a command line tool based on [Hydra](https://hydra.cc/).

Please let me know if this is worthwhile, or if you're in the process of incorporating another solution.

amykyta3, 2022-06-21:
Closing this issue since my command line tool plans diverged a bit from what I originally drafted above.
Expect an initial beta release by the end of summer.

closed 2022-06-21 by amykyta3