SystemRDL/PeakRDL #2: Announcement: Renaming all 'RALBot' projects to 'PeakRDL'

amykyta3, 2020-04-15:
I've been procrastinating this change for a while, but its about time I go ahead and implement this rename.

**What's wrong with "RALBot"?**
Although the project name had a good run, I've come to notice that the keyword "RALBot" performs pretty poorly in web searches. Apparently all major search engines redirect the search to show results for "Talbot", a breed of hunting dog, and more notably "Talbots", a retail store for women's clothing in the US. Neither search results suit this project very well.

**Why "PeakRDL"**
Short and simple. And I enjoy climbing mountains. Why not?

**How do I migrate to "PeakRDL-\*" Projects?**
In the next few days (by 2020-04-19) I will be publishing the following projects to PyPi:
* `peakrdl-html`
* `peakrdl-ipxact`
* `peakrdl-uvm`
Version numbering will continue, starting at v2.0.0

As these get published, you can simply rename your dependencies and imports to use the new prefix.

**What will happen to the old "RALBot-\*" Projects?**
On 2020-05-01, these projects will all publish one final update.
This update will include a deprecation warning pointing to this ticket.

nikitos1550, 2020-04-16:
Great! Hope new project iteration will bring us more features and customization support.

amykyta3, 2020-04-19:
All renamed projects (html, ipxact, and uvm) have now been published.

Be sure to check out the new PeakRDL-uvm. I reworked its implementation to fix several quirks. Also added some nifty features and have ideas for a few more in the future.

closed 2020-09-13 by amykyta3