SystemRDL/PeakRDL #3: Integration of other tools

krcb197, 2021-08-01:
There seem to be some other tools being developed under the PeakRDL name including:[PeakRDL-verilog](https://github.com/hughjackson/PeakRDL-verilog) 

I also have a project to develop a python wrapper generator that follows a similar structure. 

Is there a process or developers guidr for tools other people are developing, in order bring them into the PeakRDL project and command line tool.

amykyta3, 2021-08-13:
I admit I have abandoned building the command line tool for a bit due to other priorities. At my workplace, we already developed our own solution to tie intou our toolchain, so building an open-source version of a command line tool got a little neglected :pensive: 

I'm always open to having people pull in their exporters into the SystemRDL github group. If you have something that you think others would find useful and want it published in this group, let me know.

Some recommendations based on past projects I have encountered:
* Try to make your exporter follow a similar usage pattern to other PeakRDL exporters. For example, provide an exporter object with a single `Exporter.export(node, path, **kwargs)` method. This isn't necessary, but is nice when everyone follows a similar usage style that s familiar and predictable.
* Avoid overly niche or proprietary design decisions. It may be incredibly useful for your use-case, but if it isn't generic (or extensible enough) others may not find the tool very valuable.
* Documentation! Most of the PeakRDL tools I have so far are simple enough that they only need a small readme file. Something like a Python abstraction layer generator may need more details so making a readthedocs.io page may be appropriate. I'm currently working on my own [SystemVerilog generator](https://github.com/SystemRDL/PeakRDL-regblock) and that will *definitely* need a lot of documentation.
* Make sure it works! Try to anticipate the more complex or seldom used parts of the RDL language (multi-dimensional arrays, references, etc). If you don't support a feature, validate that the design doesn't use it and throw an error (unlike what I neglected to do here haha :rofl: SystemRDL/PeakRDL-html#26)

krcb197, 2021-08-13:
@amykyta3 thank you for your encouragement, that seems like a fair set of guidelines. My goal to publish a python package to generate python wrapper from systemRDL will now be extended to include integration into PeakRDL. Assuming I am satisfied it meats the criteria.

GitHub seems to suggest that you create a `contributing.md` file in the repostry `.github` folder with this information. I think it would be useful to include this there rather and close this tread,  see [Contributor Guidelines](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/setting-guidelines-for-repository-contributors)

I agreed that a good set of tests is needed to validate the potential packages, the [Accelera Generaic Example](https://github.com/SystemRDL/systemrdl-compiler/blob/47544743719450dacbd13f8a398127c911d2a6cb/examples/accelera-generic_example.rdl) has been useful to me and I have developed a few more, see [PeakRDL-Python TestCases](https://github.com/krcb197/PeakRDL-python/tree/main/tests/testcases). I have extended my test for rest conditions based on the feedback on #systemrdl/peakrdl-html#26

closed 2022-05-25 by amykyta3