SystemRDL/PeakRDL #5: AttributeError: 'dict' object has no attribute 'select'

kele14x, 2022-08-01:
Hi,

I'm trying PearRDL, but simple `peakrdl --help` will report error:

```
PS C:\Users\xxx> peakrdl --help
Traceback (most recent call last):
  File "C:\Users\Kele\AppData\Roaming\Python\Python39\Scripts\peakrdl-script.py", line 33, in <module>
    sys.exit(load_entry_point('peakrdl==0.2.0', 'console_scripts', 'peakrdl')())
  File "C:\Users\Kele\AppData\Roaming\Python\Python39\site-packages\peakrdl\main.py", line 39, in main
    Dump(),
  File "C:\Users\Kele\AppData\Roaming\Python\Python39\site-packages\peakrdl\subcommand.py", line 90, in __init__
    self.importers = get_importer_plugins()
  File "C:\Users\Kele\AppData\Roaming\Python\Python39\site-packages\peakrdl\plugins\importer.py", line 82, in get_importer_plugins
    eps = metadata.entry_points().select(group='peakrdl.importers')
AttributeError: 'dict' object has no attribute 'select'
```

I'm using Python 3.9.13 on Windows 10

amykyta3, 2022-08-01:
Thanks for reporting this @kele14x!
I hadn't gotten around to testing on all versions of Python so I missed this bug. I'm still in the process of setting up cross-platform & cross-version unit tests.

Published a fix just now. Try `v0.3.0`.

kele14x, 2022-08-01:
Hi @amykyta3,

Thanks for your fix! The issue is resolved and I can get correct `--help` output.

closed 2022-08-01 by kele14x