SystemRDL/PeakRDL #7: Why does PeakRDL require git

krcb197, 2022-08-11:
I have installed peakrdl in a clean python 3.10 virtual environment.

Then I try running:
```peakrdl --help```

I get the following error:
```
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet
```

It is not clear why Git needs to be present to run the peakrdl tools, can you explain this requirement?

amykyta3, 2022-08-17:
Thanks for reporting. It seems all my testing environments implicitly include git, so I missed this.

This message comes from the `gitpython` package which is used by `PeakRDL-html` to generate source links in the generated report. I'll figure out how to suppress this for environments without git installed.

amykyta3, 2022-08-17:
Fixed. Try again after upgrading the following package:
```
python3 -m pip install --upgrade git-me-the-url
```
The latest version suppresses this import error.

krcb197, 2022-08-17:
That has fixed the issue, thank you for looking at that for me @amykyta3

closed 2022-08-18 by amykyta3