SystemRDL/PeakRDL #8: Multiple output operations

krcb197, 2022-08-17:
In many cases multiple outputs from the same set of inputs are needed for example:
- HTML
- Registers
- Test Bench
- Header Files

Would it be possible to setup multiple output from a single set of definitions?

amykyta3, 2022-08-18:
Yeah this was actually part of the original plan I described in the project announcement (#1). Since then I opted for a subcommand-oriented approach, but I am still interested in eventually adding a `peakrdl batch` subcommand that allows you to describe multiple export operations via a YAML file or something similar. I am still experimenting with how I want the current subcommand/plugin scheme to work. Once that settles down, it will inform how I can expand these concepts to a batch command.

For now, you'd need to run separate `peakrdl` commands for each operation.

krcb197, 2022-08-18:
The compound command line allows the use of a make file to manage the process when using

KptainBiwouak, 2023-05-30:
From my point of view, as it is easy to embed these commands in a simple bash or csh scripts, I do not see the necessity to create a 'peakrdl batch' command. The user has to built his own execution script depending of his own usage, except if a specific sequence has to be followed to generate a specific output: in that case, it could be more efficient to launch the tool one time instead of several times.

amykyta3, 2023-10-24:
Closing. Not likely to implement a batch mechanism any time soon

closed 2023-10-24 by amykyta3