SystemRDL/PeakRDL #16: Migrate to `importlib_metadata` in entry point discovery

shareefj, 2023-03-20:
When you install a package in editable mode, the stdlib version of the `metadata` package reports duplicate entry points.  This causes issues for PeakRDL as the plugin manager code finds these duplicates and bails out.

The solution seems to be to use the `importlib_metadata` package which fixes the underlying issue and can be used across all versions of Python.  There is a migration guide here: https://importlib-metadata.readthedocs.io/en/latest/migration.html

amykyta3, 2023-03-20:
Could you provide more details on this issue? I use editable package installs pretty regularly during development and have not encountered this issue. I'd be interested in reproducing it and understanding the issue better.

Which version of Python are you using?
Which version of PeakRDL do you have installed?

amykyta3, 2023-03-23:
Excellent testcase. Thanks!

It seems there is some subtle difference in how `pyproject.toml`-based plugins define their entry points that affects Python 3.8 & 3.9. Since this only affects these older versions, I would rather fix my existing entry points loader rather than introducing another 3rd party dependency.

Also, looks like I'm behind the times and ought to update to use `pyproject.toml`. I didn't realize this was an official PEP recommendation.

amykyta3, 2023-03-23:
Fixed in v0.8.0.

Since your example was pinning to a pretty old version of PeakRDL, beware of the change in the plugin API introduced in v0.7.0 (#17).
See updated docs: https://peakrdl.readthedocs.io/en/latest/for-devs/exporter-plugin.html#plugin-descriptor-class

closed 2023-03-23 by amykyta3

shareefj, 2023-03-23:
Thanks.  I think `pyproject.toml` is still classed as experimental but I don't understand fully where the limitations lie.  I think `setup.cfg` and `setup.py` are safe but I've moved to using the TOML for all new packages I create.  Not an expert here at all!

shareefj, 2024-01-21:
Hi there,

I've just hit this bug again in the latest release.  I'm trying to get an editable installation of `peakrdl-regtool` and am hitting the same metadata issue.

I don't think I ever looked at your solution back when you pushed the fixes but I think you added it to the wrong branch.  The issue was with >3.10 I think.  I can currently perform an editable install on 3.9 but not on 3.10.

I think this was the original issue where I found the `importlib_metadata` solution:  https://github.com/pypa/setuptools/issues/3649

shareefj, 2024-01-21:
Having just modified my local `peakrdl` installation, if you remove the first `if` block in `entry_points.py` so that everything above 3.8.0 filters out duplicates, that seems to work.

amykyta3, 2024-01-21:
> I don't think I ever looked at your solution back when you pushed the fixes but I think you added it to the wrong branch.

Can you confirm which version of PeakRDL you are using? The fix you refer to has been merged and deployed for many months.

I just tried with 3.10 and could not reproduce any issue. If you are running with the latest PeakRDL, please give me some more detailed steps to reproduce this issue.

shareefj, 2024-01-22:
Everything is the latest from a manual `pip install peakrdl` with no version pinning.  Clone peakrdl-regblock and do an editable install into a venv.