SystemRDL/PeakRDL #17: Pip install peakrdl==0.6.0 fails

Risto97, 2023-03-20:
Hello, I just noticed a problem, because I was using PeakRDL for my plugins and I had the version of peakrdl fixed to 0.6.0 in setup.py.

The problem is that if someone installs peakrdl at tag 0.6.0, the setup will install new versions of peakrdl-ipxact,html....
And there is no ExporterSubcommandPlugin in peakrdl 0.6.0 so launching peakrdl fails.

Solution would be to change setup.py of peakrdl on tag 0.6.0 to limit the version of other packages.

There is probably nothing to fix here, I just wanted to leave it here if someone runs into the same problem, they should use the latest peakrdl and not fix to the old version.

amykyta3, 2023-03-21:
Thanks.
Starting with PeakRDL 0.7.0, I introduced some breaking changes to the plugin API to address some limitations in the original implementation. Since the other PeakRDL-* tools also needed to change in order to match the new API, if you pin PeakRDL to an older version, then it may cause an incompatibility.

I usually avoid compatibility issues like this, but since PeakRDL is in pre-production beta I am still in the phase of teh design where I may need to make more drastic changes.

Appreciate your patience as I build up this part of the toolchain!

Risto97, 2023-03-21:
I understand, its not too critical to introduce such a change at this point.
If I may add one remark maybe, did you think of separating peakrdl from all of the plugins, so they are not installed when installing peakrdl.

To illustrate my use case, I am using peakrdl-html, peakrdl-regblock and two of my plugins, but I am not using the rest of the toolchain. I am invoking the tools sometimes from python script or cmake wrappers for peakrdl.
This way the unnecessary dependencies get pulled into my virtual environment.

Another question and a advertisement for my plugins.
Do you allow for other plugin creators to keep the name scheme PeakRDL-<plugin>?

I created a C++ hardware abstraction layer driver generator PeakRDL-halcpp
https://github.com/Risto97/PeakRDL-halcpp

And a simple SVG IP block pinout generator tool, that generates IP blocks like this one, I use it to embed it into the html documentation. PeakRDL-ipblocksvg
https://github.com/Risto97/PeakRDL-ipblocksvg
[image]

amykyta3, 2023-03-23:
@Risto97
I started a separate issue #18 to discuss your first point further.

Regarding use of the `PeakRDL-` prefix, I don't mind if others use it for their own custom plugins at all. In fact, it is a good way to communicate that these are installable plugins that other community members can experiment with :smile:

amykyta3, 2023-03-23:
fyi - I started a community plugin page and added your C++ HAL. Nicely done. See: https://github.com/SystemRDL/.github/blob/main/community_plugins.md

amykyta3, 2023-05-12:
Closing since most users have likely upgraded past this issue by now

closed 2023-05-12 by amykyta3