SystemRDL/PeakRDL #18: Consider option to only install PeakRDL CLI without core plugins

amykyta3, 2023-03-23:
Splitting this into a separate request described by @risto97:

Did you think of separating peakrdl from all of the plugins, so they are not installed when installing peakrdl.

To illustrate my use case, I am using peakrdl-html, peakrdl-regblock and two of my plugins, but I am not using the rest of the toolchain. I am invoking the tools sometimes from python script or cmake wrappers for peakrdl.
This way the unnecessary dependencies get pulled into my virtual environment.

amykyta3, 2023-03-23:
In most cases I want to publish PeakRDL as a complete tool that is ready to use out-of-the-box. However I can see the value of pruning it down to a more lightweight install in some cases.

Will look into what mechanisms, if any, allow something like this.
One option would be to publish a separate package that only includes the CLI component, but I'm not too keen on breaking this project apart further as it creates dependency pains. Will look into whether there are cleaner ways to accomplish this.

Risto97, 2023-03-23:
I can propose the following solution.

Rename the current PeakRDL to something like PeakRDL-cli, that will only install the cli interface and no plugins.
Create a PeakRDL-toolchain (or a better name) for repo that only has setup.py with plugins as requirements, this way you can fix the PeakRDL-cli version to the plugins that are compatible.

Another option is to keep PeakRDL repo for toolchain and create another one for cli.

amykyta3, 2024-12-14:
Sorry for sitting on this one for so long. Mostly, I wanted to "kick the tires" a bit on the current deployment strategy and see how it scales. Initially I had thought this could be addressed using some clever use of optional dependencies.
In the end, I agree that splitting out the CLI component makes sense and scales best.

Here is what I propose:

* The Python library in this repository will be published as `peakrdl-cli`, and include no other PeakRDL plugin dependencies.
    * Plugin import paths remain unchanged and will still reference the usual `peakrdl` module by name.
* Any plugins, unofficial or official, should list `peakrdl-cli` as a dependency.
    * If the plugin also provides a non-PeakRDL API, then the CLI dependency should be listed as an optional dependency under the tag: `peakrdl-pluginname[cli]`
* This repository will continue to provide a `peakrdl` package
   * This would become a source-less package that only links the dependencies to the CLI, and other core plugins
   * Intent is to not break compatibility, and still provide an easy quick-start experience for new users.

This ends up fixing a few issues:
* Allows power-users to pick & choose a subset of core tools
* Untangles the improper dependency relationship that caused #17. In turn, this gives me more flexibility to extend the CLI plugin utilities further and reliably have plugins describe dependencies to new features.

amykyta3, 2024-12-17:
Implemented in v1.2.0

closed 2024-12-17 by amykyta3

Risto97, 2024-12-17:
Hello @amykyta3 .

Happy to see that you are kicking off the development on PeakRDL.
Its been an excellent tool that I am happy to be using on the new workplace as well.
Let me know if you decide to provide commercial support at some point😉

I think the new `peakrdl-cli` package makes sense, as to not break the flows for users relying on `peakrdl`.
Thank you for implementing this.