SystemRDL/PeakRDL #27: Generating RAL Package

GaAveMo, 2023-06-09:
PeakRDL tool generates uvm ral package not correctly. For all registers, the volatile field is always 1, it doesn't matter if I write in ip-xact volatile true or false. The tool ignores this field. And while generating SystemRDL from ip-xact is the same issue. 
I used these commands`
1. Convert other formats to SystemRDL - `peakrdl systemrdl atxmega_spi.xml -o atxmega_spi.rdl`
2. Create a UVM Register Model - `peakrdl uvm atxmega_spi.rdl -o atxmega_spi_uvm_pkg.sv`

MarinaQT, 2023-06-14:
Had already an exchange with Alex around the volatile property. My question there was about the corresponding property in RDL. Because n RDL there is donttest and dontcompare properties. Apparently you have to set these properties "by hand" or "by script" directly in the *.rdl view.
Now, i have edited our blocks to *.rdl format, and added the donttest and dontcompare properties, but the transcription to the UVM view is not being affected by these properties... I was expecting the volatile field in configure to be affected.
For example, if i set the field property in RDL to either
donttest = true¦false;
dontcompare = true¦false;

it is always resulting in the UVM field configure (an example below) with 
bit volatile = 1
bit individually_accessible = 0

**Any hints how to debug that? Or am i doing a wrong interpretation about the donttest and dontcompare properties?**

**References below:**

- example uvm output extract from a field description

        virtual function void build();
            this.fieldName = uvm_reg_field::type_id::create("fieldName");
            this.fieldName.configure(this, 16, 0, "RW", 1, 'hFF, 1, 1, 0);
        endfunction : build
        

- UVM_Class_Reference_Manual_1.2

function void configure(
 uvm_reg parent,
 int unsigned size,
 int unsigned lsb_pos,
 string access,
 bit volatile,
 uvm_reg_data_t reset,
 bit has_reset,
 bit is_rand,
 bit individually_accessible
)

amykyta3, 2023-07-20:
@GaAveMo sorry for the delay. I missed your question.

This looks like it is a side-effect of how the IP-XACT importer works. I have opened an issue here to better track this bug: https://github.com/SystemRDL/PeakRDL-ipxact/issues/21

@MarinaQT 
Agree that the missing donttest/dontcompare is an issue. Issue that tracks this limitation is here: https://github.com/SystemRDL/PeakRDL-uvm/issues/23

Also please do not hijack other people's issue tickets with unrelated discussions.  It makes it difficult to track.

closed 2023-07-20 by amykyta3

MarinaQT, 2023-08-03:
> [EXTERNAL] This sender is outside your organization.

@Alex ***@***.***> : sorry Alex, it was not my intention to mix up tickets. Was trying to follow your recommendations (in previous answer) to check this ticket.

This looks like it is a side-effect of how the IP-XACT importer works. I have opened an issue here to better track this bug: SystemRDL/PeakRDL-ipxact#21<https://github.com/SystemRDL/PeakRDL-ipxact/issues/21>

@MarinaQT<https://github.com/MarinaQT>
Agree that the missing donttest/dontcompare is an issue. Issue that tracks this limitation is here: SystemRDL/PeakRDL-uvm#23<https://github.com/SystemRDL/PeakRDL-uvm/issues/23>

Confidential and Proprietary Information. The information contained in this email message and all attachments transmitted with it is confidential information intended solely for the addressee. The information may be privileged. You are hereby notified that any dissemination, distribution or copying of any confidential or privileged information contained in this email, without our express written consent is strictly prohibited. If you have received this email in error, please immediately notify us and delete this email and any attachments from your system. Thank you.