SystemRDL/PeakRDL #29: Argument top in peak html not recognising sub addressmaps

MarinaQT, 2023-06-14:
According to documentation https://github.com/SystemRDL/PeakRDL-html it is possible to declare a internal addrmap in the RDL as top for the html output.
Tried to do so, declaring as top the Regfile addrmap instead as per default the last addrmap, which would be the Memorymap on my block.rdl file, but it was not accepted. The call and error messages are copied below:

...
top_node = 'BlocknameRegfile0'
peakrdl html --peakrdl-cfg {config_file} -t {top_node}  -o {output_file} {input_file} 

error msg
fatal: Elaboration target 'BlocknameRegfile0' not found

and double checked with "peakrdl dump" that the name is correct and recognised.

Any hints what could be the root cause?
thanks for your help and attention!

amykyta3, 2023-06-15:
The `top_node` argument needs to be a SystemRDL *type name* of the addrmap, not the *instance name*. Furthermore, the type name needs to be in the root namespace.

It is hard to know exactly since you did not provide an example, but I suspect `BlocknameRegfile0` is a child instance name.

Try using the `peakrdl globals` command - the output of this is all of the addrmap type names that are visible in the root namespace, and are valid inputs for the `-t` argument.

MarinaQT, 2023-06-16:
Hello Alex
thanks for your swift answer.
We gathered three examples, and would like to ask the following questions:

**(A) File GenericCounter_rm.rdl**
Here experimented the properties donttest and dontcompare, for their effect into the UVM output.
Unfortunately could not observe any differences when changing the values of these properties.
In the UVM output always get:
UVM_NO_COVERAGE and configure with volatile bit =1
Can you please comment which effects are already built in peakrdl to support these properties for UVM output?

**(B) File GenericCounter_rm.rdl**
In this file one field is declared as counter. But adding or removing this property had no effect on the UVM output.
It does have an effect on the regblock output. 
Can you please comment which effects are already built in peakrdl to support this property for UVM output?
For example, could be an add on to declare a callback class, where one would describe the counter behavior.

**(C) File GenericCounter_w_intr_rm.rdl**
Here tried the intr property (and removed counter), and the effect in the UVM output was only noticeable in the name of the class. But no further statements were added (related to the intr property).

**(D) File GenericCounter_w_type_rm.rdl**
Here tried to declare the Regfile_type to see if could add this level as top for "peakrdl html" command. But still it is not recognised by "peakrdl globals" as you commented above. So can not do:
 peakrdl html ../input/GenericCounter_w_type_rm.rdl -o ../output/docs/GenericCounter_rm.html -t GenericRegfile_type
 
 thanks for your attention and support
 
 [GenericCounter_examples.zip](https://github.com/SystemRDL/PeakRDL/files/11772004/GenericCounter_examples.zip)

amykyta3, 2023-06-17:
**[A]**
The current UVM generator is very limited. I am planning on doing a major rewrite of the UVM generator and will likely include better support for donttest/dontcompare.
See the issue tickets specific to UVM here: https://github.com/SystemRDL/PeakRDL-uvm/issues

**[B & C]**
UVM's register model classes are pretty limited and do not have any mechanism to encode some of the more powerful concepts that SystemRDL allows. This includes things like counters and interrupts.

**[D]**
Please read this section on how PeakRDL elaboration works: https://peakrdl.readthedocs.io/en/latest/processing-input.html#top-level-elaboration

The component type `GenericRegfile_type` is declared in an internal lexical scope, and therefore is not visible as a global type.
If you restructure your file as shown below, `GenericRegfile_type` becomes available in the root lexical scope, and is therefore a global type.

```
addrmap GenericRegfile_type {
    desc = "Generic address block";

    // (Contents not shown)
    // ...
};

addrmap GenericD__GenericMemorymap0 {
    desc = "Generic memory map";

     GenericRegfile_type GenericRegfile0;
};
```

Alternatively, you could completely remove the `GenericD__GenericMemorymap0` hierarchy since it does not seem to be adding any benefit in this example.

MarinaQT, 2023-06-17:
Thanks again for the effective exchange. A few more precisions please:

[A] Do you have a prevision for the peakrdl-uvm update (apparently planned as milestone peakrdl-uvm version 3.0)? 
      We can perhaps contribute to it, for implementing some specific planned feature or testing.  Let us know (:o)
   
[B] & [C] Do i understand right, that these RDL concepts, intr and counter, are currently only processed by peak-regblock?
What about we add an option via config file to generate tasks / functions in the field class (e.g. for counter) which can be invoked via callbacks?

[D] Clear. We have this extra hierarchy because other blocks combine Registerfile and Memoryfile. But indeed for peakrdl-regblock to work on these examples, i had indeed to remove the "extra" hierarchy.

amykyta3, 2023-06-17:
**[A]**
Not sure. I have a few other PeakRDL features I want to implement first, so it is difficult for me to know when I will work on the UVM component.

**[B & C]**
One of the improvements to the UVM generator is to allow better support for user customization. See: https://github.com/SystemRDL/PeakRDL-uvm/issues/17

MarinaQT, 2023-06-23:
Morning Alex, two more questions please.
What about considering the following 2 features in future releases?

[E] Adding basic support for hdl_path property in RDL exporter. Understand not full syntax from IPXACT is translatable, but the basic one is translatable and already useful. Cause for many this lacking translation implies extra post-processing scripting steps.

[F] What about supporting in the 'peakrdl dump' a mode which explores the hierarchical output as documented here:
https://systemrdl-compiler.readthedocs.io/en/stable/examples/print_hierarchy.html

cheers & regards

amykyta3, 2023-06-24:
[E]
RDL exporter already supports this - it will write all properties that have been loaded into the tool. The limitation is in the IP-XACT importer which does not support it.
Please see the [existing issue that has already been opened regarding this topic](https://github.com/SystemRDL/PeakRDL-ipxact/issues/14).

[F]
Please open a separate feature request ticket for this.

closed 2023-07-20 by amykyta3