SystemRDL/PeakRDL #31: [FEATURE]is there any interest to have a pyuvm exporter

sean-galloway, 2023-07-15:
**Is this related to a PeakRDL sub-command?**
If so, please open an issue in the appropriate `PeakRDL-***` project: https://github.com/orgs/SystemRDL/repositories

**Describe the problem/limitation you think should be addressed**
A clear and concise description of what the problem is.

I’m trying to stitch together a complete set of free tools capable of generating an SoC infrastructure and be able to simulate and validate it. Currently, Icarus and ventilator do not support uvm.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The cleanest solution that I see is to write an exporter for the pyuvm format. I would be willing to take on or help with this.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The obvious choice that pops into my mind is for me to create a competing tool that creates pyuvm and all of the stuff this tool does. I honestly do not mean to sound brash, but while working for a very large tech company, I have written this tool, minus the pyuvm. So, it wouldn’t be a huge leap for me to recreate it. Since I have this industry experience, I think I would be an asset to helping make PeakRDL better.

**Additional context**
Add any other context or about the feature request here.

amykyta3, 2023-07-20:
Great idea!
I personally have not used pyuvm, nor am I all too familiar with it, but have seen it grow in popularity.
If you are willing to implement and maintain a pyuvm generator, I'm sure the SystemRDL community would benefit from this.

Fair warning - I am planning to significantly refactor the PeakRDL-uvm generator in the future, so I would not recommend using that as a starting point.

sean-galloway, 2023-07-20:
Hi,

How soon is the refactor? I’m looking to create the pyuvm content as part of a YouTube series that I’m developing. However, it’ll probably be a few months before I’ll be close to working with RDL. So, I’m ok to hold off working on the pyuvm generator until (though I’ll definitely look thru the code and other generators.)

-sean

NajamKhalil, 2024-01-17:
**When we use APB4 interface for csr generation the Penable of APB interface is not getting used in the decoding logic for accessing CSRs do we have any idea regarding this?

As a reference:**

`interface apb4_intf #(

parameter DATA_WIDTH = 32,

parameter ADDR_WIDTH = 32

);

// Command

logic PSEL;

logic PENABLE;

logic PWRITE;

logic [2:0] PPROT;

logic [ADDR_WIDTH-1:0] PADDR;

logic [DATA_WIDTH-1:0] PWDATA;

logic [DATA_WIDTH/8-1:0] PSTRB;

// Response

logic [DATA_WIDTH-1:0] PRDATA;

logic PREADY;

logic PSLVERR;
`
**But in generated RTL for register decoding as access PENABLE is not used**

`always_ff @(posedge clk or negedge arst_n) begin

if(~arst_n) begin

is_active <= '0;

cpuif_req <= '0;

cpuif_req_is_wr <= '0;

cpuif_addr <= '0;

cpuif_wr_data <= '0;

cpuif_wr_biten <= '0;

end else begin

if(~is_active) begin

if(s_apb.PSEL) begin

is_active <= '1;

cpuif_req <= '1;

cpuif_req_is_wr <= s_apb.PWRITE;

cpuif_addr <= {s_apb.PADDR[5:2], 2'b0};

cpuif_wr_data <= s_apb.PWDATA;

for(int i=0; i<4; i++) begin

cpuif_wr_biten[i*8 +: 8] <= {8{s_apb.PSTRB[i]}};

end

end

end else begin

cpuif_req <= '0;

if(cpuif_rd_ack || cpuif_wr_ack) begin

is_active <= '0;

end

end

end

end

// Response

assign s_apb.PREADY = cpuif_rd_ack | cpuif_wr_ack;

assign s_apb.PRDATA = cpuif_rd_data;

assign s_apb.PSLVERR = cpuif_rd_err | cpuif_wr_err;
`
**ANY IDEA WHY?**

amykyta3, 2024-12-14:
Sorry I missed this last message - the UVM exporter rework will be mainly oriented around not using Jinja templating so extensively. I have found that for generators with a lot of conditional output templating, Jinja scales fery poorly and becomes difficult and confusing to maintain. If you want a reference, see some aspects of the regblock generator for how to partition more complex concepts in a code generator.