SystemRDL/PeakRDL #34: [FEATURE] C header file output

imerkado91, 2023-09-10:
Is it possible to add another output option for C header file (*.h)?
Or maybe there is a plugin that already does so (making C header file from SystemRDL input)?

amykyta3, 2023-09-10:
This is coming soon!
I actually started actively working on it a few days ago. Expect something in the coming weeks.

Watch this space for updates: https://github.com/SystemRDL/PeakRDL-cheader
This was originally started by @krcb197 who handed it off to me. I am working on overhauling it to add features and act as a core plugin to PeakRDL.

imerkado91, 2023-10-16:
Hi @amykyta3 ,

are there any updates about this topic?
is the C header expected to be part of the peak-RDL sometime soon?
or should I use the repository of PeakRDL-cheader for that?

amykyta3, 2023-10-16:
Very soon! Expect something official this week.

In the meantime, you can get a preview by installing directly from my dev branch:
```bash
python3 -m pip install git+https://github.com/SystemRDL/PeakRDL-cheader@dev/redesign
```
This will install the extension and make it available as `peakrdl cheader ...`

amykyta3, 2023-10-17:
Published!
Upgrade `peakrdl` to v1.0.0 and it will include the C header generator.

closed 2023-10-17 by amykyta3