SystemRDL/PeakRDL #35: [FEATURE] Enable use of environment variables and/or relative paths in -f files

mpriestleyidex, 2023-10-22:
Imagine I want to pass a long list of RDL files or include dirs to `peakrdl`, and those files are scattered around my repository.
This seems like a good fit for a `.f` file. But I'd like to specify those paths relative to either the repository root, or to the `.f` file itself e.g.:
```
-I${REPO_ROOT}/some/path
${REPO_ROOT}/another/path/thing.rdl
```
or
```
-I../../some/path
../../another/path/thing.rdl
```

Relative paths already have the behaviour of being relative to the current working directory, so probably better to use environment variables here.

amykyta3, 2023-10-24:
I like the idea.

As much as I'd love to support relative paths (relative to the .f file), I can't think of a clean way to implement it generically for the tools. The main challenge is that the .f file processor doesn't "know" what a given PeakRDL plugin's args are, and whether they are to be interpreted as file paths or simply regular strings. 

I think implementing the mechanism for environment variables to be expanded is far more possible and easier for me to maintain. Seems easy enough to tokenize them and perform the string replacement in-place.

mpriestleyidex, 2023-10-24:
Grand.
I think all you need do is call https://docs.python.org/3/library/os.path.html#os.path.expandvars on the strings you pull from the file.

amykyta3, 2023-10-26:
Implemented. Will be in next release

closed 2023-10-26 by amykyta3