SystemRDL/PeakRDL #36: How to include rtl libraries while using peakrdl where the systemrdl file has a few instantiations from those libraries?

saicharan0112, 2023-11-10:
I am new to systemRDL and I am looking at the peakrdl library to extract headers, verilog code and documentation from .rdl files. I have an rtl library which is used to create instances in the actual .rdl file. But I am not sure how to let peakrdl know about the library location. I know that `include is something more that I can use for hierarchical register definition models but I don't think that will be useful in my case. I would really appreciate if I can get some help on using peakrdl in my case, thanks!

amykyta3, 2023-11-11:
Could you provide a little more detail? I don't think I understand your question.

`include` is one way to join RDL files together, but If you are trying to compile multiple RDL files, you can also simply provide multiple in the command-line:
```
peakrdl <command> file1.rdl file2.rdl file3.rdl
```

saicharan0112, 2023-11-11:
sure, I have a file named `uart.rdl` inside a folder say `rtl_lib` and I have two levels of hierarchies where I created an instance of the uart in my rdl file at level 2 say `block2.rdl`. I used `include` to build the hierarchy in systemrdl (and peakrdl is not complaining about that) but it complains about the instance that I created. 

But I just tried the method you proposed by listing all rdl files beside the main ones and it no longer gives me that instantiation error. Thanks for the help!

closed 2023-11-11 by saicharan0112

amykyta3, 2023-11-11:
Great! I'm glad it worked out.

When listing files for the command-line tool, be sure to list them in dependency-order - child objects first, then top-level last.

saicharan0112, 2023-11-11:
Thank you very much. I will remember this as further explore this tool.