SystemRDL/PeakRDL #51: Style guide recommendations

amykyta3, 2025-03-12:
fyi @jeras Moving discussion to an issue.

> Thanks for the feedback!
> 
> 1. Completely agree. Preserving original source doc is important
> 
> 2. Parameters are good if they control multiple properties, but DPAs are definitely easier for quick changes. Also some things are impossible to describe unless you use DPAs (interrupt masking, or other cross-field behavior). I'll still add as a recommendation
> 
> 3. I hesitate to make this a general rule. I routinely work on large SoCs that are fully described in RDL that have more complex hierarchies than this. Even so, I think there are some more generalized recommendations I can make regarding sensible use of hierarchy.
> 
> 4. Agree. Sometimes I see designs that use unnecessary prefixes in front of every single thing. 

 _Originally posted by @amykyta3 in [de57816](https://github.com/SystemRDL/PeakRDL/commit/de578160b578cc964968b16efef076fc14d6eb53#commitcomment-153596281)_

closed 2025-03-12 by amykyta3

jeras, 2025-03-12:
Regarding the hierarchy. Something I would tell users to avoid is mixing `regfile` and `reg` within the same parent. While it is supported by the standard, it looks out of place.

amykyta3, 2025-03-12:
Definitely disagree.
Using hierarchy is an extremely useful way to use structure to describe intent to the reader. Mixing reg/regfile is something I have done many times. Restricting this would be inappropriate.

Consider this simple example of a register block for a hypothetical IPv4 packet filter:
```systemrdl
addrmap ip_filter {
    default sw = rw;
    default hw = r;

    regfile ipv4_rule {
        regfile ipv4_header {
            reg ctrl_reg #(longint unsigned W=1) {
                field {} value[W] = 0;
            };
            ctrl_reg #(.W(3)) flags;
            ctrl_reg #(.W(8)) protocol;
            ctrl_reg #(.W(32)) src;
            ctrl_reg #(.W(32)) dst;
        };

        reg {
            field {} enable;
            field {} clear_stats;
        } ctrl;

        reg {
            field {
                sw = r;
                hw = na;
                counter;
            } n;
        } hit_count;
        hit_count.n.hwclr = ctrl.clear_stats;

        ipv4_header value;
        ipv4_header mask;
    }

    reg {
        field {} enable;
        field {} bypass;
        field {} drop_all;
    } ctrl;

    ipv4_rule filter_rules[32];
};
```

I didn't even include any description strings, but you can probably infer a lot about how this device would operate.
Notice how the `ctrl` registers intuitively exist next to the grouped registers inside regfiles, and imply their purpose. Flattening this to a shallow structure, or wrapping things in more regfiles would make readability worse.

The purpose of a style guide is to steer designers to use a common coding dialect, not to restrain them from expressing themselves. Setting stringent rules on hierarchy would be inappropriate for a style guide intended for a broad audience.

jeras, 2025-03-12:
Thanks.

I often think of the IP protocol header as a great example for SystemVerilog packed structures. I agree, `regfile` does provide a way to group registers within the same peripheral.

I also agree not mixing `regfile` and `reg` should not be a policy, still this approach is common in Microcontroller/SoC documentation I remember.

The reason for my dislike comes from my research on generating documentation.
In a document every `regfile` parent is documented with a table containing a list of `children`. Mixing `regfile` and `reg` affects the consistency of such tables. Many documents list `reg` reset values (sometimes also access type) in such tables, but this values can't be applied to `regfile`. So a table generator has the option to not include reset/access_type values in `regfile` children tables (I choose this one) or to leave some table cells empty (not consistent).

commit b199a0f7 (2025-03-12): Style guide updates. closes #51