SystemRDL/PeakRDL #53: How to make peakrdl-regblock-vhdl peakrdl-markdown work in an Python Venv?

rafaelnp, 2025-05-12:
I created a Python virtual environment on Arch Linux, and installed:

```sh
pip install peakrdl-regblock-vhdl[cli] peakrdl-html peakrdl-markdown
Collecting peakrdl-html
Using cached peakrdl_html-2.11.0-py3-none-any.whl.metadata (5.1 kB)
Collecting peakrdl-markdown
Using cached peakrdl_markdown-1.0.2-py3-none-any.whl.metadata (2.9 kB)
Collecting peakrdl-regblock-vhdl[cli]
Using cached peakrdl_regblock_vhdl-1.0.0.3-py3-none-any.whl.metadata (3.8 kB)
Collecting systemrdl-compiler<2,>=1.29.0 (from peakrdl-regblock-vhdl[cli])
Using cached systemrdl_compiler-1.29.3-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.3 kB)
Requirement already satisfied: Jinja2>=2.11 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-regblock-vhdl[cli]) (3.1.6)
Collecting peakrdl-cli>=1.2.3 (from peakrdl-regblock-vhdl[cli])
Using cached peakrdl_cli-1.3.0-py3-none-any.whl.metadata (1.4 kB)
Requirement already satisfied: antlr4-python3-runtime<4.14,>=4.11 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from systemrdl-compiler<2,>=1.29.0->peakrdl-regblock-vhdl[cli]) (4.13.2)
Requirement already satisfied: colorama in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from systemrdl-compiler<2,>=1.29.0->peakrdl-regblock-vhdl[cli]) (0.4.6)
Requirement already satisfied: markdown in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from systemrdl-compiler<2,>=1.29.0->peakrdl-regblock-vhdl[cli]) (3.8)
Requirement already satisfied: typing-extensions in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from systemrdl-compiler<2,>=1.29.0->peakrdl-regblock-vhdl[cli]) (4.13.2)
Requirement already satisfied: git-me-the-url>=2.0.3 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-html) (2.1.0)
Requirement already satisfied: python-markdown-math in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-html) (0.9)
Requirement already satisfied: gitpython<4.0.0,>=3.1.41 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-markdown) (3.1.44)
Collecting peakrdl<2.0.0,>=1.1.0 (from peakrdl-markdown)
Using cached peakrdl-1.3.0-py3-none-any.whl.metadata (2.9 kB)
Requirement already satisfied: py-markdown-table<0.4.0,>=0.3.3 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-markdown) (0.3.3)
Requirement already satisfied: pygments<3.0.0,>=2.15.0 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-markdown) (2.19.1)
Requirement already satisfied: setuptools<66.0.0,>=65.5.1 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from peakrdl-markdown) (65.7.0)
Requirement already satisfied: gitdb<5,>=4.0.1 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from gitpython<4.0.0,>=3.1.41->peakrdl-markdown) (4.0.12)
Requirement already satisfied: smmap<6,>=3.0.1 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from gitdb<5,>=4.0.1->gitpython<4.0.0,>=3.1.41->peakrdl-markdown) (5.0.2)
Requirement already satisfied: MarkupSafe>=2.0 in /home/rnp/tmp/peakhdl/lib/python3.13/site-packages (from Jinja2>=2.11->peakrdl-regblock-vhdl[cli]) (3.0.2)
Collecting peakrdl-ipxact<4,>=3.4.1 (from peakrdl<2.0.0,>=1.1.0->peakrdl-markdown)
Using cached peakrdl_ipxact-3.5.0-py3-none-any.whl.metadata (43 kB)
Collecting peakrdl-regblock<2,>=0.19.0 (from peakrdl<2.0.0,>=1.1.0->peakrdl-markdown)
Using cached peakrdl_regblock-1.0.0-py3-none-any.whl.metadata (2.2 kB)
Collecting peakrdl-systemrdl<2,>=0.3.0 (from peakrdl<2.0.0,>=1.1.0->peakrdl-markdown)
Using cached peakrdl_systemrdl-0.3.0-py3-none-any.whl
Collecting peakrdl-uvm<3,>=2.3.0 (from peakrdl<2.0.0,>=1.1.0->peakrdl-markdown)
Using cached peakrdl_uvm-2.3.0-py3-none-any.whl
Collecting peakrdl-cheader<2,>=1.0.0 (from peakrdl<2.0.0,>=1.1.0->peakrdl-markdown)
Using cached peakrdl_cheader-1.0.0-py3-none-any.whl.metadata (42 kB)
Using cached peakrdl_regblock_vhdl-1.0.0.3-py3-none-any.whl (91 kB)
Using cached systemrdl_compiler-1.29.3-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (11.0 MB)
Using cached peakrdl_html-2.11.0-py3-none-any.whl (5.6 MB)
Using cached peakrdl_markdown-1.0.2-py3-none-any.whl (18 kB)
Using cached peakrdl-1.3.0-py3-none-any.whl (14 kB)
Using cached peakrdl_cheader-1.0.0-py3-none-any.whl (41 kB)
Using cached peakrdl_ipxact-3.5.0-py3-none-any.whl (40 kB)
Using cached peakrdl_regblock-1.0.0-py3-none-any.whl (85 kB)
Using cached peakrdl_cli-1.3.0-py3-none-any.whl (31 kB)
Installing collected packages: systemrdl-compiler, peakrdl-uvm, peakrdl-systemrdl, peakrdl-regblock-vhdl, peakrdl-regblock, peakrdl-ipxact, peakrdl-cli, peakrdl-cheader, peakrdl-html, peakrdl, peakrdl-markdown
Successfully installed peakrdl-1.3.0 peakrdl-cheader-1.0.0 peakrdl-cli-1.3.0 peakrdl-html-2.11.0 peakrdl-ipxact-3.5.0 peakrdl-markdown-1.0.2 peakrdl-regblock-1.0.0peakrdl-regblock-vhdl-1.0.0.3 peakrdl-systemrdl-0.3.0 peakrdl-uvm-2.3.0 systemrdl-compiler-1.29.3
```

 and cannot run the peakrdl-regblock-vhdl and peakrdl- markdown on the peakrdl  atxmega_spi example:

```sh
❯ peakrdl markdown atxmega_spi.rdl -o html_doc
usage: peakrdl [-h] [--version] [--plugins] [-f FILE] [--peakrdl-cfg CFG] <subcommand> ...
peakrdl: error: argument <subcommand>: invalid choice: 'markdown' (choose from dump, globals, preprocess, regblock-vhdl)

❯ peakrdl regblock-vhdl ./atxmega_spi.rdl -o regblock --cpuif axi4-lite
usage: peakrdl [-h] [--version] [--plugins] [-f FILE] [--peakrdl-cfg CFG] <subcommand> ...
peakrdl: error: argument <subcommand>: invalid choice: 'regblock-vhdl' (choose from dump, globals, preprocess, systemrdl, uvm, c-header, regblock, ip-xact, html)
```

the packages installed in the virtual environment:

```sh
❯ pip list
Package                Version
---------------------- -------
antlr4-python3-runtime 4.13.2
colorama               0.4.6
git-me-the-url         2.1.0
gitdb                  4.0.12
GitPython              3.1.44
Jinja2                 3.1.6
Markdown               3.8
MarkupSafe             3.0.2
peakrdl                1.3.0
peakrdl-cheader        1.0.0
peakrdl-cli            1.3.0
peakrdl-html           2.11.0
peakrdl-ipxact         3.5.0
peakrdl_markdown       1.0.2
peakrdl-regblock       1.0.0
peakrdl-regblock-vhdl  1.0.0.3
peakrdl-systemrdl      0.3.0
peakrdl-uvm            2.3.0
pip                    25.1.1
py-markdown-table      0.3.3
Pygments               2.19.1
python-markdown-math   0.9
setuptools             65.7.0
smmap                  5.0.2
systemrdl-compiler     1.29.3
typing_extensions      4.13.2
``

The html generation works:

```sh
❯ peakrdl html  atxmega_spi.rdl -o html_doc
❯ ls -l
Permissions  Size User Date Changed     Date Created   Name
.rw-r--r--  2.3Ki rnp  2025-05-12 21:50 2025-05-12 21:50 atxmega_spi.h
.rw-r--r--  2.5Ki rnp  2025-05-12 21:50 2025-05-12 21:50 atxmega_spi.rdl
.rw-r--r--  8.0Ki rnp  2025-05-12 21:50 2025-05-12 21:50 atxmega_spi.xml
.rw-r--r--  3.2Ki rnp  2025-05-12 21:50 2025-05-12 21:50 atxmega_spi_bf.h
.rw-r--r--  4.1Ki rnp  2025-05-12 21:50 2025-05-12 21:50 atxmega_spi_uvm_pkg.sv
drwxr-xr-x      - rnp  2025-05-12 22:33 2025-05-12 22:33 html_doc
drwxr-xr-x      - rnp  2025-05-12 22:34 2025-05-12 21:50 regblock
```

I could not figure out, what is the problem, any suggestions are welcome.

amykyta3, 2025-05-12:
Check that the `peakrdl` executable is being used from your virtual environment, and not some other installation path:
```
which peakrdl
```
I suspect you have two conflicting installations.

You can also confirm which plugins are loaded via:
```
peakrdl --plugins
```

rafaelnp, 2025-05-12:
That was the case thanks. As I am not so well versed on python, I thought pipx would not interfere with the virtual environment.

```sh
❯ pipx list                                                                                                                                                                                                                                                                    
venvs are in /home/rnp/.local/share/pipx/venvs
apps are exposed on your $PATH at /home/rnp/.local/bin
manual pages are exposed at /home/rnp/.local/share/man
   package peakrdl-regblock-vhdl 1.0.0.3, installed using Python 3.13.3
    - markdown_py
    - peakrdl
    - pygrun
```

I ran 

```sh
❯ pipx uninstall-all                                                                                                                                                                                                                                                           
uninstalled peakrdl-regblock-vhdl! ✨ 🌟 ✨

❯ pipx list                                                                                                                                                                                                                                                                    
nothing has been installed with pipx 😴
```

and it worked, thanks.

```sh
❯ peakrdl html  atxmega_spi.rdl -o html_doc   ; tree -d html_doc                                                                                                                                             
html_doc
├── content
├── css
├── data
├── fonts
│   ├── FontAwesome
│   ├── Lato
│   └── RobotoSlab
├── js
└── search

10 directories

❯ peakrdl regblock-vhdl atxmega_spi.rdl -o regblock/ --cpuif axi4-lite ; tree regblock 
regblock
├── atxmega_spi_pkg.sv
├── atxmega_spi_pkg.vhd
├── atxmega_spi.sv
└── atxmega_spi.vhd

1 directory, 4 files
```

If I may piggyback this issue: is there a way to install the aforementioned python packages with pipx, so I do not have to create and manage a virtual environment?  I tried so far many combinations and none of them worked (thus, there was the `peakrdl-regblock-vhdl` remaining installation  I had to clean)

rafaelnp, 2025-05-14:
After a couple of attempts, I found a way around the virtual environment, at least for the daily usage. Given the already created virtual environment, I created a symlink passing the absolute path to peakrdl to $HOME/bin, that is in my $PATH and I can run peakrdl. It should not cause problems, and the virtual environment. It may be useful to anyone trying to avoid using virtual environments for non-python development workflows...

```sh
❯ peakrdl --plugins
importers:
        ip-xact --> peakrdl-ipxact 3.5.0
exporters:
        c-header --> peakrdl-cheader 1.0.0
        html --> peakrdl-html 2.11.0
        ip-xact --> peakrdl-ipxact 3.5.0
        regblock-vhdl --> peakrdl-regblock-vhdl 1.0.0.3
❯ peakrdl regblock-vhdl atxmega_spi.rdl -o regblock --cpuif axi4-lite  
❯ peakrdl html atxmega_spi.rdl -o html_dir 
❯ tree regblock ; tree html_dir -d
regblock
├── atxmega_spi_pkg.sv
├── atxmega_spi_pkg.vhd
├── atxmega_spi.sv
└── atxmega_spi.vhd

1 directory, 4 files
html_dir
├── content
├── css
├── data
├── fonts
│   ├── FontAwesome
│   ├── Lato
│   └── RobotoSlab
├── js
└── search

10 directories
```

closed 2025-05-14 by rafaelnp

amykyta3, 2025-05-14:
Yeah I was about to recommend the same. `pipx` implicitly creates a separate isolated environment for each package installed. In order for plugins to be visible, they need to be inside the same isolated environment. `pipx inject` would be my recommendation rather than hacking things with symlinks.