SystemRDL/PeakRDL #55: Support for multiple memory maps in an IP-XACT file

cyrusbaz, 2025-06-04:
Hello. Thank you for such a great program. :)

Background:

For our project we are reading in multiple IP-XACT files from different sources/vendors. Our goal is to produce a memory map and then a UVM-RAL, DOCX, PDF, etc. Your program is great, super-fast, and produces very efficient results. My current tool chain (using Magillem) combines these various XML and produces a memory map only XML output. Problem is the output is really large.

Problem:

The issue I am hitting is that there are 2 memory maps in the IP-XACT. I have one that is accessed via APB, and the other, which is accessed via a different, proprietary interface. When I generate HTML only the second memory map documentation is produced (HBM_CM0P_SBM_RING_IN_MM). Same if I add in the peakrdl-docx plugin and create a DOCX.

**Describe the solution you'd like**
Support for multiple memoryMaps. It would be great to be able generate documentation for all, or some of them, by specifying the name. Here is a snippet of the IP-XACT I have:

    <spirit:memoryMaps>
        <spirit:memoryMap>
            <spirit:name>APB_SLV_MM</spirit:name>
            ....
        </spirit:memoryMap>
        <spirit:memoryMap>
            <spirit:name>HBM_CM0P_SBM_RING_IN_MM</spirit:name>
           ....
        </spirit:memoryMap>
    </spirit:memoryMaps>

It would be cool if PDF was also added as an output.

**Describe alternatives you've considered**
I can manually edit the IP-XACT file to make two different IP-XACT, one for each memoryMap.

**Additional context**
Add any other context or about the feature request here.

amykyta3, 2025-06-04:
This is somewhat of an intentional effect. RDL semantics effectively require there to be one and only one root node in an elaborated design.

Since the IP-XACT effectively describes two separate devices, the IP-XACT importer will transcribe these into two different top-level addrmap definitions. More details on how that works here: https://peakrdl-ipxact.readthedocs.io/en/latest/importer.html

During the elaboration, the compiler will blindly choose the last one it sees by default, unless you specify an alternate one using the `--top type_name` command line flag. Using `peakrdl globals <ipxact file>` will list the available type names.

If you want to stitch them together into one top-level, you can do so using a top-level "glue" SystemRDL file and instantiate the components that were reported by `peakrdl globals`:

```js
addrmap top {
    component_name__APB_SLV_MM apb_slv_mm @ 0xA000_0000;
    component_name__HBM_CM0P_SBM_RING_IN_MM @ 0xA000_1000;
};
```

Compiling this along with your IP-XACT will let you merge them into one monolithic register tree (as well as assign the global address offsets)

cyrusbaz, 2025-06-04:
Thank you! That makes complete sense.

How about PDF output on the wish list? :)

amykyta3, 2025-06-04:
Worth mentioning - Another option is to export to an intermediate format like docx or Markdown using one of the community plugins, and then convert to PDF: https://peakrdl.readthedocs.io/en/latest/community.html#documentation

cyrusbaz, 2025-06-04:
Thank you very much for all the information. I will try these out. I look forward to using PeakRDL more.