SystemRDL/PeakRDL #62: Question on writing plugin and using toml file as entry point

kpakya123, 2025-09-17:
Hello,
I saw the video: https://www.youtube.com/watch?v=ag9TaeeWlzs from Alex Mykyta. Thanks for the detailed demo.
I am trying to write a plugin (exporter), and running into what seems to be a very baseic error. Pardon my lack of Python expertise.
As Per the peakrdl doc and video i extended plugin class:
`import argparse
from peakrdl.plugins.exporter import ExporterSubcommandPlugin
from systemrdl.node import AddrmapNode, RegNode

class SVParamsPkgExporter(ExporterSubcommandPlugin):
  short_desc = "Generates pkg containing param for each reg, reg-field and related ..."

  def __init__(self, name):
    self.name = name
    
  def add_exporter_arguments(self, arg_group: argparse.ArgumentParser) -> None:
    arg_group.add_argument("pname_pre", default='', help="prefix to add to pkg name")

  def do_export(self, top_node:AddrMapNode, options:argparse.Namespace) -> None:
    pass
`

I put a toml file with contents:

`[peakrdl]

python_search_paths = [
                    "<>/prjsimtools/lib/lib_pytools"
]

plugins.exporters.SVParamsPkgExporter = "peakrdl_sv_params_pkg_plugin:SVParamsPkgExporter"
`

I am seeing the foll. error
`<>/prjsimtools/peakrdl.toml: error: peakrdl.plugins.exporters.SVParamsPkgExporter: module 'peakrdl_sv_params_pkg_plugin' has no attribute 'SVParamsPkgExporter'`

I thinking i am missing something very basic. I have less understanding of how plugins work.
I just put a "pass" in body of do_export() to first see if my plugin shows up in peakrdl --help.

Appreciate your help.

Thanks,

kpakya123, 2025-09-18:
Hello Alex,

Here is an update.

I moved the plugin source from where it was , and put it in the current working directory (from where the peakrdl is invoked)
This time the python runtime found the attribute in the plugin source module.  I am now able to see my exporter listed in "peakrdl --help"
peakrdl.toml file: 

```
[peakrdl]

python_search_paths = [
#                    "/india/proj_brute/brute/work/pkhatri/w_001/prjsimtools/lib/lib_pytools"
                     "."
]

```

Here is the peakrdl --h output:

```
peakrdl --help
peakrdl --help
usage: peakrdl [-h] [--version] [--plugins] [-f FILE] [--peakrdl-cfg CFG] <subcommand> ...

PeakRDL is a control & status register model automation toolchain.

For help about a specific subcommand, try:
    peakrdl <command> --help

For more documentation, visit https://peakrdl.readthedocs.io

options:
  -h, --help           show this help message and exit
  --version            show program's version number and exit
  --plugins            Report the PeakRDL plugins, their versions, then exit
  -f FILE              Specify a file containing more command line arguments
  --peakrdl-cfg CFG    Specify a PeakRDL configuration TOML file

subcommands:
    dump               print register model contents to stdout
    globals            list all globally accessible types that can be elaborated as top
    preprocess         Preprocess SystemRDL and write the result to a file
    regblock           Generate a SystemVerilog control/status register (CSR) block
    c-header           Generate a C header definition of an address space
    ip-xact            Export the register model to IP-XACT
    systemrdl          Export the register model to SystemRDL
    html               Generate HTML documentation
    uvm                Generate a UVM register model
    SVParamsPkgExporter
                       Generates pkg containing param for each reg, reg-field and related ...
```

I will now proceed with writing the exporter.

Many thanks for your help.

amykyta3, 2025-09-18:
Nice! Glad you got it to work.

kpakya123, 2025-10-02:
Thanks Alex.